chili-epfl/chilitags #87: marker configuration for a cube ?

flobotics, 2016-05-21:
hi,
i try to build a marker configuration for a cube. The cube has on 5 sides, only the bottom is free, because there is a clip, chilitags.  Every side of the cube is 2x2 cm, so the chilitag is 2x2cm, but the black-square inside is only 13x13mm, to have a white surrounding for detection.

Now i dont get how to define it ? Lets say we got this cube [http://goinkscape.com/wp-content/uploads/2014/09/rubiks-13.png](url)  and the yellow side is side-nr.1.

On side-nr.1, we got 9 yellow cubes, where only the center-cube is the black-square of the chilitag, and the rest of the yellow-cubes are the white surrounding.

So when the first marker in the configuration file has: translation [0.0, 0.0, 0.0] and rotation [0.0, 0.0, 0.0] what does the second marker has as configuraton ?

If we go clockwise, so the green-side of the cube is side-nr.2, we got  translation [0.0, 0.0, -20.0] and rotation [+90.0, 0.0, 0.0]    (or a negative rotation ?)

is there anywhere a cube configuration as example, or are there anythings that could help me in designing this config file ?

thanks

flobotics, 2016-05-22:
i got a config for a cube with 5 tags, but its not really correct. If someone knows it, please help

here the config:
http://discourse.ros.org/t/robotic-humanoid-hand/188/36

qbonnard, 2016-05-22:
Hello,

Sorry for answering too little, too late... Looks like you managed to make some progress anyway. I think the missing piece of information in https://github.com/chili-epfl/chilitags/blob/master/share/tag_configuration_sample.yml is that the 0 of markers re the top left corners. That explains the 3.5 offset you have.

Your configurations seems in good shape... What makes it "not really correct" ?

As for the tool that could help you designing the config file (again, too little, too late, sorry), I'd say that the 3d samples could help (c++ or javascript, if you don't want to compile), with some tricks :
- set `keep: 1` to see individual markers compared to the object they compose
- disable filtering and other niceties to get the raw data
- try and add markers one by one in the config file, show only two to the camera, and hide/show the one you're trying to add to see whether the object "jumps". If so, it means the configuration is not correct.

Not sure if that's clear... If not, and if it's not too late, I'll try to be more reactive !

severin-lemaignan, 2016-05-23:
Hello!

I've added an example of a cube layout (SVG) + the configuration. Cf #88 

Hope it helps!

flobotics, 2016-05-23:
thanks for help, but the cube-layout and config is confusing me :) 

- in the SVG you write "size 30mm", what is meant with this? the size of the complete side (with white borders?) or only the black-square ? I think the complete side is 30x30mm, or? 

- why do you directly use translation and rotation for the first tag, why not start with trans: and rot: = [0.,0.,0.]   for simplicity ? 

- in the config you use "size: 30", but then you translate it by 15.0  and very confusing 25.9 ? why do you translate it by the half of the tag-size ?  position zero is left-upper side of the tag, or?

in the marker_configuration_sample.yml  is written "marker" but one must use "tag" instead to work, also in the Readme.md of the creator are wrong sizes for the zoom for the examples, just a note

thanks

flobotics, 2016-05-24:
when i dont use keep:1 in any tag, where does the tf frame is projected to ? Always to the first tag described in one object ? Also if i turn the cube around 180 degrees in the image stream ?

Is there a way to use the center of the cube as tf-frame?

i tried to print the svg file, but all tools had problems with the transparency.

qbonnard, 2016-05-24:
Hello,

Here are the answers I can provide, and I'll let @severin-lemaignan answer the others:

> when i look with keep:1, it seemed that it uses the black square inside the white ?

That's right, the tag is the black border and its content.

> does every trans and rotation depend on the first tag in the myobject1: definition, or on the tag before the last tag definition ? so like chain-dependency ?

None of the above ;). Each `object` has an implicit referential on its own, and the rototranslations of the tags are defined against it. So if you define one of the tags as translated and rotated by a null vector, the referential of this tag and the objects are the same. But you don't need to do that, and do as Séverin did for example, where each tag on the side of the cube has a rotation and translation w.r.t. the center of the cube. Your approach was to have one of the faces as origin plane. Both work with chilitags, but you approache might yield a more complex configuration.

> in the marker_configuration_sample.yml is written "marker" but one must use "tag" instead to work

Hum, which version are you talking about ? The latest on github is fine, but it indeed was like that on earlier versions...

> also in the Readme.md of the creator are wrong sizes for the zoom for the examples

thanks, it should be fixed now

> i tried to print the svg file, but all tools had problems with the transparency.

Even Inkscape ? I think that's what Séverin used.

> when i dont use keep:1 in any tag, where does the tf frame is projected to ?

Wherever the origin of the object is, based on the pose of the tags detected and their transformation relative to the object (specified by translation and rotation). Keep has not much to do here, it's mostly a debugging option. If set to 1, instead of having only my_object in the list of detected objects, you'll also have "virtual objects" for each of the kept tag, named tag_## 

> Always to the first tag described in one object ?
If and only if the first tag has translation : [0,0,0] and rotation [0,0,0]

> Also if i turn the cube around 180 degrees in the image stream ?
Yep

Hope that helps !

flobotics, 2016-05-24:
thanks, but i still got a problem. I printed the cube severin made and use his config file, but when i turn the cube around so that  6,7,9,10 tag is to see, the tf frame is always jumping into the opposite direction of the camera-frame, then when it sees the tag more, the tf-frame jumps back into the center of the cube ?

When i then turn the cube, so that top is to see, and back, the tf-frame turns into the opposite direction. But only on every second tag, with the other tags, the tf-frame turns into the same direction as the cube.?

qbonnard, 2016-05-25:
@severin-lemaignan is this cube layout something you've used, or something you made up as an example ?

severin-lemaignan, 2016-05-25:
I've used it for a quick demo, but I may well have one rotation wrong :-) I'll try to check!

flobotics, 2016-05-26:
i got a correct config now with a center tf-frame, thanks alot @qbonnard  and @severin-lemaignan for your help

closed 2016-05-26 by flobotics

qbonnard, 2016-05-26:
Good to hear, and you're welcome !
Did you fix Séverin's svg/yaml ? Would you mind sharing your cube configuration to save your struggle to the next adventurer in Chilitagsland ? ;)

flobotics, 2016-05-27:
no, i didnt fixed severins svg/yaml, i build it directly for my cube. My cube has side-length of 20x20millimeter, and the black-border of the tag is 13x13millimeter. A picture of the cube is here: [http://discourse.ros.org/t/robotic-humanoid-hand/188/37](url)  and the config is:

distal_phalanx:
     - tag: 15
       size: 13
       translation: [-6.5, -6.5,-10.]
       rotation: [0., 0., 0.]
     - tag: 3
       size: 13
       translation: [10., -6.5, -6.5]
       rotation: [0., -90., 0.]
     - tag: 4
       size: 13
       translation: [ 6.5, -6.5, 10.]
       rotation: [ 0., -180., 0.]
     - tag: 2
       size: 13
       translation: [-10., -6.5, 6.5]
       rotation: [0., 90., 0.]
     - tag: 14
       size: 13
       translation: [-6.5, -10., 6.5]
       rotation: [-90., 0., 0.]

@qbonnard i reduced my cpu-load to around 60% and get 30fps, while running chilitags, but the tf-frame is flickering, even if the cube is standing still and the picture is all the time the same? could this be, because of the small 13x13mm tags? but when it can easily detect it and it does not move, why does the tf-frame move/flicker?

qbonnard, 2016-05-27:
Thanks!

Looking at your video, my guess yould be that the flickering comes from the tracking. Is it enabled ? There are other tweaks you can play with to get a detection closer to what you want, especially the filtering, and making sure that the image is not resized by chilitags before being processed. But admittedly, the tags are pretty small on the image :) 13mm would be fine if the camera were closer of course.

Does the tf frame also jump when the cube does not move (and is close from the camera), and you hide tags one by one ? If so, there is probably still something wrong in the configuration.

Also, how about having a strip of chilitags on the faces of the cube instead of just one ? Actually, a single (long enough) plane with a strip of chilitags could be quite good at detection the rotation too.

flobotics, 2016-05-29:
i bought a logitech c525 webcam, now i can use the chilitags samples, i only used the ros_markers before, which seemed to only use estimate: auto foundObjects = chilitags3d.estimate(inputImage);

When i now look into the samples directory, there are async-detection detect-from-file  detect-live  estimate3d  estimate3d-gui  filter3d-gui and tracking. 

- What sample is for what intention?  

- how can i specify my markers-config.yml, i only found out that you can use estimate3d-gui -c markers.yml   but the other samples seemed not to have this option?

- With which sample can i best figure out if my cube-config is correct ?

qbonnard, 2016-05-30:
> What sample is for what intention?
- async-detection detects tags in a background thread
- detect-from-file is the simplest detection example getting images from a file
- detect-live is the simplest detection example getting images from a webcam
- estimate3d and estimate3d-gui are the same, but with 3d estimation instead of simple detection
- filter3d-gui shows the filtering options (to smooth little detection errors)
- tracking tracks markers once they have been detected to speed up processing

They are meant to document the usage of the library with their code and comments

> how can i specify my markers-config.yml
You'd have to modify a sample for your needs (IIRC, it's just about adding a call to `chilitags3D.readTagConfiguration(configFilename);`)

> With which sample can i best figure out if my cube-config is correct ?
estimate3d-gui would allow you to see the object "jump" when you hide a marker that is not correctly placed