chili-epfl/chilitags #89: camera calibration

flobotics, 2016-05-26:
hi,
the only text about camera calibration i could found is on ros_markers site: images and camera calibration are read from a standard ROS camera

What does this mean ? The roslaunch command only uses the image-topic and the camera-frame-topic, no camera_info topic ? Or is there anywhere a config file ?

I use a kinect2, but i think i got no good camera calibration in chilitags for this, or am i wrong ? (auto-calibration )

qbonnard, 2016-05-26:
I don't know about ROS, but if you want to know everything and more, here is OpenCV's related documentation : http://docs.opencv.org/2.4/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html

On a more efficient note, if you want to generate your camera calibration to be used by chilitags, you can simply use the output of opencv's sample in ` opencv_source_code/samples/cpp/calibration.cpp`

Hope that helps

severin-lemaignan, 2016-05-27:
@flobotics What is your actual question? how to calibrate your camera? how to use calibration data with chilitags?

ROS uses a specific mechanism that 'hides' the calibration details in a `camera_info` topic. But unless you want to use chilitags with ROS (via `ros_markers`), this is not very important.

flobotics, 2016-05-30:
hi @severin-lemaignan ,
my question was , how to use camera calibration with chilitags ? i bought now a logitech c525 to have the more standard way, i use usb_cam which uses the camera calib.yml file and i think will publish it on camera_info topic. So the question about calibration is not actual anymore, thanks. But i got another question, why is the tracking with ros_markers much more flickering than the tracking with chilitags directly ? Are there other default options set ?