chili-epfl/chilitags #91: IDs detected, that don't exist anymore.

Martin-Oehler, 2016-09-14:
I'm using a very basic setup where I feed single images (batch) into Chilitags::find using DETECT_ONLY. After detection I draw the found tags into the image and display them.

I noticed that a tag, which was found in the first image, is still detected at the same position in the second image, even if it is not present anymore in the second image.

qbonnard, 2016-09-14:
Hi Martin,

Did you call `chilitags::setFilter(0, 0);` ? By default, tags "persist" 5 for frames before disappearing.

Martin-Oehler, 2016-09-15:
Hello qbonnard,
thanks for your advice, I did indeed not call `chilitags::setFilter(0, 0);`. 

I think this is a strange default behavior since it only makes sense for tracking and is really surprising when you use `DETECT_ONLY` for batch processing.

qbonnard, 2016-09-15:
The rationale for this default behavior is that in it's simplest use (instantiating Chilitags, then calling `find` right away), detection should do the "right thing" for most users (nothing new here). We thought that most users provide a feed of images from the camera in a robotic or AR context, and the right thing was to have tag not appear and disappear too often... Even if it means a latency 5 frames to realize that a tag has disappeared. Out of curiosity, may I ask what's your use case ?

So I assume I can close this issue ?

Martin-Oehler, 2016-09-15:
It always depends on the application I guess.

I'm calibrating a robot arm by detecting the tags from multiple poses and minimizing the error to their expected position. You can find the fork [here](https://github.com/tu-darmstadt-ros-pkg/robot_calibration), if you are interested.

Sure, you can close it now, thanks for your help.

closed 2016-09-15 by qbonnard