chili-epfl/chilitags #95: Chilitags3D.estimate(cv::Mat) Doesn't work against opencv 3.3/3.2

Mizerb, 2017-12-11:
When run, get an error on assertion in opencv /modules/core/src/matrix.cpp at line 2542 with: 

` CV_Assert( mtype == type0 || (CV_MAT_CN(mtype) == 1 && ((1 << type0) & fixedDepthMask) != 0) );`

I understand these versions of opencv may not be supported, I just wanted to make sure it's written down somewhere.

severin-lemaignan, 2017-12-11:
@Mizerb I might be wrong, but it really looks like you're passing an empty image to chilitags. Can you double check your image is not empty?

Mizerb, 2017-12-11:
@severin-lemaignan Morning, thanks for taking a look at this. I'm pretty sure the image wasn't empty. I ran cv::imshow against the mat, and it showed the correct input. Also ran chilitags::Chilitags find() on the mat, which worked successfully with opencv 3.3/3.2. 

I'll check again later, and it might be something else on my end. It just when I changed out the version to 3.1 it began to work.

severin-lemaignan, 2017-12-11:
@Mizerb ok, then it's not due to an empty image... interesting...

MikeGameDev, 2017-12-18:
Hi, I had this same problem in the "estimate3d-gui" sample. It seems that OpenCV is complaining that one matrix has a type of CV_32FC1 while the other has a type of CV_64FC1. I was able to fix this by converting both matrices to CV_64FC1 prior to calling cv::Rogrigues.

`template<typename RealT>
void EstimatePose3D<RealT>::operator()(std::string const& name,
                                       std::vector<cv::Point3_<RealT> > const& objectPoints,
                                       cv::Mat_<cv::Point2f> const& imagePoints,
                                       typename Chilitags3D_<RealT>::TagPoseMap& objects)
{
    // Find the 3D pose of our tag
    cv::solvePnP(objectPoints, imagePoints,
                 mCameraMatrix, mDistCoeffs,
                 mTempRotation, mTempTranslation, false,
#ifdef OPENCV3
                 cv::SOLVEPNP_ITERATIVE);
#else
                 cv::ITERATIVE);
#endif
    //TODO: Rotation and translation vectors come out of solvePnP as double

	**mTempRotation.convertTo(mTempRotation, CV_64FC1);
	mTempTranslation.convertTo(mTempTranslation, CV_64FC1);**

    if(mFilter3DEnabled)
        mFilter3D(name, mTempTranslation, mTempRotation);

    cv::Rodrigues(mTempRotation, mTempRotMat);

	 

   objects[name] = {
        (RealT)mTempRotMat(0,0),    (RealT)mTempRotMat(0,1),    (RealT)mTempRotMat(0,2),    (RealT)mTempTranslation.at<double>(0),
        (RealT)mTempRotMat(1,0),    (RealT)mTempRotMat(1,1),    (RealT)mTempRotMat(1,2),    (RealT)mTempTranslation.at<double>(1),
        (RealT)mTempRotMat(2,0),    (RealT)mTempRotMat(2,1),    (RealT)mTempRotMat(2,2),    (RealT)mTempTranslation.at<double>(2),
        0,                          0,                          0,                          1
    };
}`